longmetal7127/Infinite-Recharge #23: Make color sensor work

EarlJr53, 2020-02-06:
In the code, the implementation of the color sensor is commented out. At some point, we need to figure out how to properly implement it.

i-am-gizm0, 2020-02-06:
Maybe a `spinTo` method that takes an enum (or just a `Color`). Maybe if it isn't that color right now, it spins and if it is the right color, it doesn't spin (it could return a boolean representing if the color is the one detected).
*Some pseudocode to make my point. This isn't a real language, but I guess you could say it's influenced by Python and JavaScript.*
```
function spinTo(enum color):
  if (not isColor(color)):
    spin()
    return false
  else:
    stop()
    return true
```

Similar to that, a `rotatedSpinTo` that figures out what color to spin to for the FMS's sensor to read the correct color (it would call `spinTo`, but pass it a different value).
```
function rotatedSpinTo(enum color):
  return spinTo(rotatedColor(color))
```

For rotating a specific number of times, we could have two methods, `initRotate(int)` and `updateRotate()`. `initRotate` would take an `int`, for the number of turns. It would store the initial color value and create a counter for how many times that color has passed. `updateRotate` would check if the color changed and if it did, it would check if the new color is the initial color and if so, update the counter. If the counter is less than 2x the number of turns to rotate (because there are two of each color on the dial), it would rotate the control panel and if it has turned enough, it would not spin it. The method could return a boolean indicating whether it is done or not.
```
function initRotate(int turns):
  instance var setRotations = turns * 2
  instance var accumulated = 0
  instance var lastColor = currentColor()
  instance var initColor = currentColor()
```
```
function updateRotate():
  var currentColor = currentColor()
  if (currentColor != lastColor):
    lastColor = currentColor
    if (currentColor == initColor):
      accumulated++
      if (accumulated >= setRotations):
        stop()
        return true
  spin()
  return false
```

This is just the first thing that came to my mind and I don't have access to vscode right now, so if you don't want to do it this way, you don't have to; this is just a suggestion

fjfur10, 2020-02-11:
sam i made them strings and not enums

enums make me cry

- jon

closed 2020-02-11 by fjfur10