longmetal7127/Infinite-Recharge #40: Get Target Distance from Limelight

i-am-gizm0, 2020-02-16:
Limelight has a [great article](http://docs.limelightvision.io/en/latest/cs_estimating_distance.html) about getting the distance to a target using a limelight. @EarlJr53's `ShooterFormula` expects a distance in meters so we can use this to feed it values we need.

i-am-gizm0, 2020-02-20:
Important for the calculations:
On our bot, the lens of the Limelight is ~11in above the ground (thanks @EarlJr53)

EarlJr53, 2020-02-20:
@i-am-gizm0 the limelight camera is no longer mounted at 0 degrees, as they angled the whole shooter. I will find a value for that angle and add it to this comment.

EarlJr53, 2020-02-20:
In addition to getting the distance value, we need to get the data to align the drivetrain to the target. This was mentioned in #34 , but as I clean up the Issues list, I want to add it to here.

EarlJr53, 2020-02-25:
The alignment, as mentioned in the previous comment, is done, apart from some fine tuning. The shooter distance readings' first draft is done, and I hope to test it once the shooter is back on the robot.

EarlJr53, 2020-02-26:
I just implemented this into <master>, so we will test this afternoon, and then it should be good!

closed 2020-02-26 by EarlJr53