longmetal7127/Infinite-Recharge #44: Fix Subsystem Status Problems

i-am-gizm0, 2020-02-19:
> Right now, enabling and disabling subsystems through Shuffleboard isn't working - regardless of the value of the `SendableChooser`, they will throw `SubsystemDisabledException`s.
### What I know about the problem:
- The `SendableChooser`s are working. When you change its value, the boolean sent to Shuffleboard updates to the correct value.
- It *could* be a problem with `SubsystemManager`, but far more likely is the Subsystem class not handling the updates correctly.
### How to troubleshoot:
- [ ] Try enabling different combinations of subsystems. My theory is that the `enabled` variable belongs to `Subsystem` and is not being inherited to belong to other classes and it gets overwritten with each update (the last subsystem determines whether or not *everything* is enabled).

EarlJr53, 2020-02-20:
This seems to work today? I didn't really change anything? idk, I will continue to look into it

EarlJr53, 2020-02-21:
At this point, the enablers seem to function as intended, except when they are disabled, lots of errors are thrown in the console. Maybe if we could try to mitigate those errors, we would be all set for this issue.

i-am-gizm0, 2020-02-21:
If errors are being printed for only the ones that are disabled, it's working properly. Maybe we can have it only print once per disable (when you disable a subsystem it prints once)

EarlJr53, 2020-02-22:
Yeah that makes sense. We can go in and change that pretty easily.

EarlJr53, 2020-02-22:
Should be done!

closed 2020-02-22 by EarlJr53