longmetal7127/Infinite-Recharge #49: Get Pneumatics Working

EarlJr53, 2020-02-21:
I spent a lot of the day yesterday (2/20) writing the code for the pneumatics. I did a little bit of testing (didn't work) but that was before I finished writing it. At this point, all of the code for the compressor, the 2 climb solenoids, and the DoubleSolenoid for the Control Panel should be written, it's just a question of whether it works or not. Hopefully I can test it tonight, but I will be potentially looking for help from @i-am-gizm0 if I can't seem to get it working, as I have pretty much exhausted my online resources.

i-am-gizm0, 2020-02-22:
It should work now, but we still need to test it

EarlJr53, 2020-02-22:
I'm gonna say that this is closed for now, and if it doesn't work, we can re-open it

closed 2020-02-22 by EarlJr53