longmetal7127/Infinite-Recharge #51: Fix delay from limit switch

EarlJr53, 2020-02-21:
The limit switch works as intended... but the .delay() function freezes the entire robot for that period of time, not just the transporter. So during those 3-5 seconds that the transporter should be running (and does), we cannot drive the robot or interact with other subsystems. This needs to be fixed.

i-am-gizm0, 2020-02-21:
Instead of using [`delay`](https://github.com/LongMetal-Robotics/Infinite-Recharge/blob/2b77779011c3b8690ef8f45ed101f7f1c9f93c34/src/main/java/org/longmetal/subsystem/Intake.java#L44), you should create a timer object and `reset` it every time the limit switch is tripped. Then you can `get` the time (maybe have a method that is called every `periodic`) and if it is more than the time to run the transport, stop.

EarlJr53, 2020-02-22:
I basically coded it like that, except instead of resetting the timer, I just take a time from the timer, and then continue taking value and subtracting the initial value from them until the difference is the amount of time I want

closed 2020-02-22 by EarlJr53