longmetal7127/Infinite-Recharge #53: Come up with ideas for **Auto Code**

EarlJr53, 2020-02-22:
We need to come up with a direction we want to pursue for auto code. I am plenty happy to write it, but I'm not good at coming up with ideas. Think of this like #22 , as a thread for ideas, conversations with the rest of the team, etc.

So if y'all could comment some ideas here, I can get to writing it :)

EarlJr53, 2020-02-24:
Basic plan:
- Leave initiation line
- Shoot 3 starting balls
- Find and collect 3-5 more balls
- Shoot balls

i-am-gizm0, 2020-03-04:
In 10c333f I added two methods to drive the robot straight for a set distance

EarlJr53, 2020-03-10:
As of right now, <Auto> has 4 modes:
- Do nothing
- Drive back
- Drive back, aim, and shoot
- Drive back while collecting, aim, and shoot

Next, we need to get the Pigeon IMU and start writing auto modes in PathWeaver so that we can have more complex modes. Also, these modes need to be tested (the shooter isn't engaging for some reason).