longmetal7127/Infinite-Recharge #56: Implement Intake-End Vision Tracking

EarlJr53, 2020-02-26:
Sorta similar to #41 , but for the other end of the robot. @i-am-gizm0 already tried out the vision processing software on the ShowBot, but we now need to decide how we would like to implement it on the comp bot. If I have some time over the next few days, I will load the software onto the Pi and test it with the comp bot itself. We also need to write code to process the information that we get from the software and communicate with the Pi. I would probably defer to @i-am-gizm0 on the code side and I will stick to the software for now.

EarlJr53, 2020-02-26:
Addendum: Please work on this in the <Vision> branch!

i-am-gizm0, 2020-03-03:
I've developed a GRIP pipeline to generate contours for power cells (on the x360), but we need to put the Raspberry Pi in the robot to run this