longmetal7127/Infinite-Recharge #57: LED Strips (and panel??)

EarlJr53, 2020-02-26:
I know this isn't the most pressing issue, but it's important to get done nonetheless. If I remember correctly, @i-am-gizm0 already wrote most of the code for the Pi to control the LEDs, but if not, @fjfur10 can probably help with the Python. In the next 2 days, I'm planning on cleaning up all the wiring in the robot, and while I'm at it, I will mount the strips inside the drivetrain. At that point, all we need is to hook up the Pi and complete the code for it.

i-am-gizm0, 2020-02-28:
Firstly we need to figure out what we want to do with the LEDs, including how we want they placed on the robot and what designs we want on them. Once we decide on that, we can write code to tell the Pi what to do.

i-am-gizm0, 2020-02-28:
The source code for the LEDs on the Pi is in [LongMetal-Robotics/PiLED](https://github.com/LongMetal-Robotics/PiLED). I'll document the ~~API~~ there

i-am-gizm0, 2020-03-09:
Until we figure out the power problem (batteries draining too fast, robot browning out after not a long time), I'm going to close this issue and this will move to very near the bottom of our priorities. Once we resolve those, however, we can 1) increase the speed (#62) and 2) add LEDs

closed 2020-03-09 by i-am-gizm0