longmetal7127/Infinite-Recharge #62: A bunch of improvements

i-am-gizm0, 2020-03-06:
There are a couple of things we need to fix:
- [ ] Shooter idle: the shooter should idle at 1000RPM when it isn't doing anything else but currently it just stops
- [ ] Singulator idle: the shooter should idle spinning backward slowly to keep the power cells out of the shooter
- [ ] At some point, we should also try to increase the speed of the robot because it's a little slow right now.

EarlJr53, 2020-03-06:
I fixed both idles, the singulator and the shooter, and we can test them after school today. The drivetrain issues should also be worked on today.

i-am-gizm0, 2020-03-10:
The robot perpetually thought the climb was up so it was limiting its speed. It's a bit faster now

closed 2020-03-10 by i-am-gizm0