longmetal7127/Infinite-Recharge #63: New things to add

i-am-gizm0, 2020-03-06:
Some things that we need to add at some point:
- [ ] PID for alignment: Lining up using the Limelight isn't super accurate since it won't turn much when it is already close. We should have PID for lining up with the Limelight
- [ ] Run the shooter backwards: The power cells get stuck in the shooter because the singulator idles forward (in #62), so we should let it run backward to get it out

EarlJr53, 2020-03-06:
I added a control to run the shooter + singulator backwards manually. I also added an automatic reverse function for the drivetrain to shoot into the lower port.

i-am-gizm0, 2020-03-08:
I am somewhere near halfway through tuning the PID loop so I'll just close this issue

closed 2020-03-08 by i-am-gizm0