longmetal7127/Infinite-Recharge #71: LED Ideas

i-am-gizm0, 2020-03-10:
This is a low-priority thread, mostly so we can brainstorm ideas for LED stuff.
Don't work on this if you have something more pressing to do.

i-am-gizm0, 2020-03-10:
Something we could do (that would take a little tuning) would be for LEDs on each side of the drive train to match with the speed the motors are running at so that if we're driving forward, they run at the same speed the robot is moving relative to the ground so it looks like they're standing still. It would take a bit of effort but it'd look really cool and unique

EarlJr53, 2020-03-10:
> Something we could do (that would take a little tuning) would be for LEDs on each side of the drive train to match with the speed the motors are running at so that if we're driving forward, they run [...]

I would definitely like to take a look at that if I have some free time sometime soon. Another thing we can do is mirror some of our ShuffleBoard boxes to the LEDs, including potentially the RPMInRange or something similar. Also I want to try to put a vertical strip up the outer climb tube, so that we can do some cool stuff there.